RyanteckLTD/RTk #4: License question

lurch, 2016-04-29:
I just noticed that `LICENSE` says:
"Permission is hereby granted, free of charge, to any person obtaining a copy
of this software and associated documentation files (the "Software"), to use the
Software for educational or non commercial purposes."

Does the [Kickstarter campaign](https://www.kickstarter.com/projects/ryanteckltd/rtkgpio-a-usb-raspberry-pi-compatible-gpio-header) not count as commercial purposes? ;)
Does this mean we can't use the RTK.GPIO board for commercial purposes?

Some clarification would be great...

lurch, 2016-04-29:
ping @whaleygeek too

ryanteck, 2016-04-29:
Hi,

Ive been given permission from David for commercial use so I need to re word the license a little.

Non commercial in this case is for it to not be sold for any money. 

Thanks -
Ryan

whaleygeek, 2016-04-29:
I'm happy for Ryan to bundle a source version of a derivative work of anyio with his Rtk.GPIO board, and to sell his board, as long as he doesn't sell the software or restrict my rights in the original copyrighted software. If someone wanted to suggest some licence modifications/clarifications to the RTk Licence file, I can review them and sign them off for you. You can always put a copy of my signoff in the licence, and I can also add some clarification to the original anyio licence if you like.

lurch, 2016-04-29:
Ahhh, so it's a case of "buy the RTK board and get the software for free" ? ;-)
IANAL, so wouldn't know what modifications to suggest.

What about the firmware that comes pre-flashed on the RTK boards - does that fall under the same license as the PC-side python code, or does/will it have a separate (hopefully more liberal) license?

ryanteck, 2016-04-29:
I'll have a re-write of a license to state this. Other bits stay the same such as the no warranty.

The firmware side will be in a another repo and will be under either GNU GPL, BSD, MIT or a license similar to David's saying use of it is allowed apart from commercial. (Commercial being if someone makes a board and uses the software, not commercial environments).

lurch, 2016-04-29:
IMHO the best licence for something as central to the RTK board as the firmware, would be a well-known one (like GPL/BSD/MIT) with no non-commercial exclusions, just to avoid any confusion.
If you're worried about competitors, you could always keep the firmware repo private until you actually start shipping the KS rewards. Or perhaps you could say that the firmware is only licensed to run on boards manufactured by Ryanteck?
(as long as the PC<->RTKboard interface is 'open', allowing people to write their own implementations of it if they really wanted to)

But of course at the end of the day the licensing decisions are down to the original authors :)

ryanteck, 2016-05-01:
We're now happy with the new one so I'll mark this as closed.

closed 2016-05-01 by ryanteck