RyanteckLTD/RTk #12: GPIO 0 & 1 Are not wired up

ryanteck, 2016-05-01:
@lurch Pointed out that the odd add-on may use GPIO 0 & 1 pins which are located on the HAT EEPROM Pins.

Currently these are not wired up and the original plan was for them to not be.

The MCU may have one or two pins free. The question is weather they should be used as GPIO or not.

ryanteck, 2016-05-01:
Well after double checking the Raspberry Pi specifications say to **NOT** use these pins for anything but an EEPROM even on any Non Hat Board. 

> The ID_SC and ID_SD pins must only be used for attaching a compatible ID EEPROM. **Do not use ID_SC and ID_SD pins for anything except connecting an ID EEPROM, if unused these pins must be left unconnected**
[https://github.com/raspberrypi/hats#new-add-on-boards-basic-requirements](Hat Specification Page)

I'll see if it's possible to route these new pins on the MCU along with the other changes I need to make. However I'm going to say this is lower priority than the other bits I need to fix and route.

lurch, 2016-05-01:
It's obviously only IMHO, but it'd be a shame if the RTK board didn't support all 28 of the GPIO pins on the RPi header.

ryanteck, 2016-05-01:
Software patch released. Just need to see if I can route them.