RyanteckLTD/RTk #13: Add analogue API

ryanteck, 2016-05-01:
The MCU has a built in ADC that can be used on most pins.

While not strictly RPi.GPIO it may be nice to enable the use of these as a selling point of being used with a Pi.

lurch, 2016-05-01:
Nice idea, if it doesn't add too much extra work.
If possible, IMHO the library for this should be in a separate `anyio.ADC` module rather than incorporating it into `anyio.GPIO`, so as to not break the "swapability" between `anyio.GPIO` and `RPi.GPIO`.

Are there plans to add PWM-output? https://sourceforge.net/p/raspberry-gpio-python/wiki/PWM/

ryanteck, 2016-05-01:
Yep, thanks for reminding me to add an issue for that.

As for how I'll implement it I'll have a look as the serial port is setup under the GPIO Class which wouldn't work.

I may implement it as GPIO Not quite sure.

lurch, 2016-05-01:
Maybe you could have 'private' methods in anyio.GPIO (i.e. starting with an underscore), and then anyio.ADC could import anyio.GPIO and wrap these 'private' methods to then expose the 'public API' with different method names? \*shrug\*

lurch, 2016-05-02:
...I guess an alternative approach, would be to have a 'base layer' anyio.BaseSerial class which does all the serial communications with the RTK-GPIO board, and then you could have anyio.GPIO, anyio.ADC, anyio.I2C, anyio.SPI, anyio.UART etc. all sitting 'on top' of the shared anyio.BaseSerial class.

I added examples of separate classes for I2C, SPI and UART, because IMHO these are all features that it'd be nice for the RTK-GPIO board to support, but they aren't features included in the RPi.GPIO library, so IMHO they shouldn't be included in the anyio.GPIO library.

whaleygeek, 2016-05-02:
I always had a view that these would *not* live under GPIO, because they are not GPIO related, and there is no 'standard' abstraction for them that would be modelled.

I also had the view that each different service, such as GPIO, ADC, PWM, NeoPixel etc etc, would have a separate .c module on the target side, with an associated python abstraction on the python side. Indeed if you look at the base code, you'll see that the simple textual protocol has the facility to use different prefix letters to the commands, which are then routed to a different module handler.

If you look in the firmware folder, you'll see the beginnings of other bits and pieces that were intended to be extra linked target modules

https://github.com/whaleygeek/anyio/tree/master/anyio/arduino/firmware

Also note, I had a discussion with Ryan about this, with the addition of perhaps one extra static factory method in the static redirector, you could have multiple instances of the GPIOClient(SerialPort) abstraction and easily instantiate more than one GPIO board.

The correct way to model what you are talking about here I think, is to mirror the modular firmware structure on the python side. Thus the GPIOClient and the wire protocol in 'protocol.py' should probably be renamed as just Client, as it will then host multiple uses on top (GPIO, ADC, PWM etc). but you need a single class instance to manage the serial connection to the associated firmware, otherwise you'll get in a pickle, especially if the host application starts to use threads.

Also note the embedded 'net' transport, I always had a view that I would offer a 'GPIO over the network' service by configuring the transport to 'net' rather than 'arduino', and then using a raspberry pi network server at the other end to accept incoming connections and process the actual GPIO. This becomes easier with the RTk board, because you could write python on one Pi, remote it over the network and put a RTk board on the other pi to actually 'do' the GPIO stuff.

I think all the architecture you need is there already, but just consider renaming some of the GPIO bits as they will now transfer more than just GPIO commands to the target firmware. But please do use the module prefixing and routing on the wire protocol that is already there. (see below, there are two ways to effect GPIO, either with a pin number prefix for simplicity, or use the 'G' module prefix. So you could add 'A' for ADC, 'P' for PWM, 'S' for SPI, 'I' for I2C (in fact that is how I plan to add them to the base anyio eventually anyway).

static void command(char cmdch, char paramch)
{
  switch (cmdch)
  {
    case 'V': /* Version */
      Serial.println(VERSION_STR);
    break;
    
    case 'G': /* GPIO */
    {
      while (Serial.available() == 0)
      {
        /* busy wait for parameter to arrive */
      }
      /* GPIO commands are two chars */
      char pinch = paramch;
      char cmdch = (char)Serial.read();
      gpio(pinch, cmdch);
    }
    break;
    
    default:
      /* All other commands are rejected in this simple version */
      error(ERROR_UNKNOWN_COMMAND);
    break;
  }
}

lurch, 2016-05-02:
WRT a 'standard abstraction' for PWM, I'd assumed that anyio.GPIO would be emulating the PWM functions built into RPi.GPIO https://sourceforge.net/p/raspberry-gpio-python/wiki/PWM/  (of course how that gets implemented in terms of the low-level wire protocols is a different matter entirely).

And regarding a 'net' transport for anyio, have a look at https://github.com/RPi-Distro/python-gpiozero/issues/278 and https://github.com/RPi-Distro/python-gpiozero/issues/279

ryanteck, 2016-05-02:
Ok thinking about it.

PWM out would be aimed to be the same to the user as PWM on the pi is.

I'll agree with ADC, I2C and SPI separately on the python side. You may find ADC implemented in the GPIO class this week just for me to get it programmed in to demo for the KS and attract more backers then split it off after.

ryanteck, 2016-05-05:
A basic implementation has been done and works under the gpio class.

Next job is to move this to its own class.