RyanteckLTD/RTk #16: RTk Licence (and AnyIO Licence)?

dglaude, 2016-05-21:
Dear Ryan,

I understand you are using anyio as a base for your code.

While this as the benefit to exist, it has a very strange software licence [https://github.com/whaleygeek/anyio/blob/master/LICENSE](url) that might be very restrictive and far from compatible with other licence: 
1. For educational or non commercial purposes.
2. You agree to not restrict my rights in the use of this Software in any way. 

Do you have a formal agreement with the original copyright owner?

What does that mean that the software needed for using the RTk and what does that mean for any software that use the library?

What about the firmware running in the board, what is the licence for that one.

Could you clarify the licence status of your project?

Regards.

ryanteck, 2016-05-21:
Hi,

We were actually talking about it at issue #4  and have permission from David which is now at https://github.com/RyanteckLTD/RTk/commit/096ac117bb0c4e0c4b211a816f977f974ce7aa86

The MCU will be under a more open license (GNU GPL V3 I think) and the method of communicating will be documented a bit more so if you're not happy you can make your own library.

Is this a bit clearer?

whaleygeek, 2016-05-21:
Yes, there is agreement. I'm just checking a couple of things, but I'm considering widening the anyio licence to MIT in the future, which would resolve these queries immediately, and also mean that the code and it's derivatives could have a further reach.

dglaude, 2016-05-21:
I would just suggest a licence not incompatible with the GPL... especially, do avoid usage limitation and non commercial.

dglaude, 2016-05-22:
I have read #4 and I can confirm that "No commercial use" make this licence incompatible with GPL.

Because other can explain that better than me, let's repeat that "non-commercial clause" in a licence mean this is nor an Open Source licence nor a Free Software licence:
[https://www.quora.com/What-is-the-best-free-for-non-commercial-use-license-to-use-for-an-open-source-project](url)

The "_all modifications are re-published_" is typical, but does not say under witch licence the modified code need to be re-publish. So it does not have the "viral" effect of "must be publish under the same licence".

Then "_and you do not restrict the original author or any authors rights in any way._"
and  "_You agree to not restrict my rights in the use of this Software in any way._"
make no sense to me... The original author cannot be deprived right he has on the original software, but does he tries to acquire **for himself only** additional right on the derivative software?

All in all, I have no clue what licence this can be compatible with and what licence to use for something that one would like to distribute and must link to this code to work.

There are various "MIT" licence [https://en.wikipedia.org/wiki/MIT_License](url) but they are very compatible (with GPL) as they do not impose much. However, it does not give you the guarantee that derivative work will stay in the open. Code under that kind of licence can fall on the closed side of software and all the added feature would be lost for the common.

I believe you should investigate what he really want and what existing licence already do that rather than inventing yet another licence. All those existing licence (BSD, MIT, GPL, ...) have carefully chosen wording for a reason.

Not all hope is lost, if the protocol is sane and it is not too hard to rewrite something that implement the client side of it, then Free or Open Source software can be written **from scratch** to use that kickstarter hardware.

If the kickstarter hardware can be reprogram, then maybe other protocol are possible that would open to other compatibilities. I am in particular thinking of pigpio as this is already a client/server model (over named socket or network connection) to drive GPIO. It as a text based protocol that could be spoken on a serial link. It is supported by GPIOZero, so compatibility with this powerful Pi library can be easy.

ryanteck, 2016-05-22:
Well the hardware is using my own code which is fully open source. The fact that the API is the same means that you shouldn't need to re-program it.

You can re-program it if you buy hardware but then this won't be officially supported.

I need to have a look at pigpio as its new to me. (I've only heard about it last week) and it may be where I move to that from the start.

As for the license it now has a clause saying commercial use is allowed? I need to remove the non commercial part from it which I forgot to remove last time.

ryanteck, 2016-05-22:
I've closed this as the license is what it is. This library allows commercial use now. The only clause is you can't sell it which is the same as pretty much any Open Source License.

It is still open source and the code the board runs on is GNU GPL as I wrote that from scratch to interface with this software.

dglaude, 2016-05-22:
For me, there is only one definition of Open Source License and it is recommended not to say "Open Source" when it is not [https://opensource.org/faq#avoid-unapproved-licenses](url).

But I agree with you, the license is what it is, it is up to you and the other author to decide.
I let you with your studies and your Kickstarter, you can always come back to this issue later.

ryanteck, 2016-05-22:
Technically it is open source, The source code is there and open.

However I'll look at getting the license changed to a proper one which is looking like MIT for this and the rest is GNU GPL V3.

I'm also setting up a forum quickly which will be better suited for discussions like this.

ryanteck, 2016-05-22:
And fixed.

closed 2016-05-22 by ryanteck