RyanteckLTD/RTk #25: Cause MacOS panic when plug to MacBook Pro via USB

bowenfeng-dev, 2017-03-19:
System: macOS Sierra v10.12.3
MacBook Pro (Retina, Mid 2012)
Intel Core i7 2.7 G
16GB 1600MHz DDR3

When I plug the Rtk.GPIO into my MacBook Pro the screen blacks out and after a while it shows the system is getting problem and needs to be restarted.

ryanteck, 2017-03-19:
Hi,

Can you open this as a support thread on our forum as 1) The github is no longer used for the software (I moved to bitbucket). And 2) This is a driver issue not a python library.

Thanks

ryanteck, 2017-03-19:
Update: This is a known issue with the chipset if you install the makers drivers of the CH340 IC. We've got updated instructions over at https://rtkgpio.xyz/t/mac-install-guide/103 and there's working drivers over at https://github.com/adrianmihalko/ch340g-ch34g-ch34x-mac-os-x-driver

closed 2017-03-19 by ryanteck

whaleygeek, 2017-03-19:
Just for completeness, I think this is a Mac serial driver problem, not a RTK.GPIO board problem - I have the same problem sometimes using the micro:bit serial port with my Mac (it black screens).

bowenfeng-dev, 2017-03-20:
Thanks for the quick reply. The driver solved the problem.